geeekpi/upsv3 #1: Unable to communicate with UPS

tingles2233, 2020-04-30:
Hi, 

Ran the script with my geeekpi ups.. I have ensured I2C is enabled...
Any suggestions or anything else I am missing? Running Rasbian
pi@raspberrypi:~/upsv3 $ sudo python3 demo.py
Traceback (most recent call last):
  File "demo.py", line 20, in <module>
    read()
  File "demo.py", line 10, in read
    ina.configure()
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 175, in configure
    self._max_expected_amps)
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 305, in _calibrate
    self._calibration_register(calibration)
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 351, in _calibration_register
    self.__write_register(self.__REG_CALIBRATION, register_value)
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 383, in __write_register
    self._i2c.writeList(register, register_bytes)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 127, in writeList
    self._bus.write_i2c_block_data(self._address, register, data)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 364, in write_i2c_block_data
    self._device.write(data)
OSError: [Errno 121] Remote I/O error

tingles2233, 2020-04-30:
May have just figured this out.. Are there two versions with USB-C? I thought the new one was the only one with USB-C.. I feel I have an older model..

roeste, 2020-05-15:
> May have just figured this out.. Are there two versions with USB-C? I thought the new one was the only one with USB-C.. I feel I have an older model..

Today I received my geeekpi ups. Tried this piece of code and do have eactly the same result. BTW, I'm using an RPi 3b+ unit.

@tingles2233, you mentioned you've figured it out. Does that mean you have solved it in your case? If so, what was your solution?

buncopter, 2020-06-07:
I've had a look around, and there are 2 versions at least of the USB-C output. 1 version will have only 4 pins, and the most recent version has 6 pins. 

The 4 pin version looks like this: https://i.ebayimg.com/images/g/6IwAAOSwnexerbcD/s-l1600.jpg
The 6 pin version looks like this: https://i.ebayimg.com/images/g/rmEAAOSwUvBeqnA~/s-l1600.jpg

The 4 pin version has RTC implemented (I am not sure if correctly as I do not have the board) but no voltage/current measuring board components. 

However, if you refer to my other issue I opened, the board has no documentation of how to set RTC, or help writing script for soft shutdown on low voltage etc. I just bought this board, but with lack of documentation I cannot recommend anyone to purchase it until there is better documentation/support. For a bit more money, there are much better boards out there that come with good amount of documentation, support and sample scripts to help with implementation of the RTC/voltage/current meters.

tingles2233, 2020-06-07:
Completely agree with everything said.

I can only say one word and I am extremely happy with it PiJuice.

dejavue1985, 2021-01-09:
> Hi,
> 
> Ran the script with my geeekpi ups.. I have ensured I2C is enabled...
> Any suggestions or anything else I am missing? Running Rasbian
> pi@raspberrypi:~/upsv3 $ sudo python3 demo.py
> Traceback (most recent call last):
> File "demo.py", line 20, in
> read()
> File "demo.py", line 10, in read
> ina.configure()
> File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 175, in configure
> self._max_expected_amps)
> File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 305, in _calibrate
> self._calibration_register(calibration)
> File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 351, in _calibration_register
> self.__write_register(self.__REG_CALIBRATION, register_value)
> File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 383, in __write_register
> self._i2c.writeList(register, register_bytes)
> File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 127, in writeList
> self._bus.write_i2c_block_data(self._address, register, data)
> File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 364, in write_i2c_block_data
> self._device.write(data)
> OSError: [Errno 121] Remote I/O error

I had the same issue but was able to solve it. In my case i just had to turn on die I2C Interface of the pi. After that the demo worked fine.

yoyojacky, 2021-03-16:
make sure you installed it according to the product wiki.
and there is new one called UPS Plus can do it better: https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136

closed 2021-03-16 by yoyojacky