geeekpi/upsv3 #3: RTC and Low voltage auto shutdown

buncopter, 2020-06-04:
Hi, 

1) We don't know anything about the RTC. Can you please provide code to program it for date and time?

2) Can you please provide a script for auto shutdown (software) when low power?

For example, the X728 (competing UPS hardware) already have extensive software writeup to use the new hardware for software voltage-based shutdown, programming of RTC module, battery voltage and percentage reading, AC power loss detection etc. Just wondering if we can have something similar for the UPS V3?
https://github.com/geekworm-com/x728/blob/master/x728.sh

Thanks! Hardware is all there, just need the software side of things

serantes, 2020-06-26:
I have the same problems and If you can't detect power shutdown this hardware is a waste of money. I tried the sample code with power up and power down and results are similar.

yoyojacky, 2021-03-16:
Please try this :https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136

closed 2021-03-16 by yoyojacky