geeekpi/upsv3 #8: Read line voltage / charge status?

MaxVonEvil, 2021-08-01:
Hi, I'm guessing this UPS only has one INA219, i.e. it is not possible to read the input voltage on the UPS?
I would like to do that in order to trigger a graceful-shutdown script after X number of minutes, that is unless there is a way to intrepret the numbers from the internal INA219 as to if a power drop event has happened.

I guess one could manually add one on the charger side of the UPS? If anybody has any notes on that, I'm all ears.
On that note, can anyone explain what the "repair board" does exactly? And if one should add a second INA129, should it be placed before or after the repair board?

samuelegunawan, 2022-02-06:
As far as I understand it, the repair board forces the UPS hat to read the voltage from the battery when it's charging every 10 minutes. This is quite important because the UPS hat can supply more power than it can take, which means the battery is decreasing as you use it while connected to the wall wart.

If you don't use the repair board, then the UPS hat will show the battery at its last voltage reading before being charge. When the power from the wall wart is cut off, the UPS hat will die immediately because it is out of power, even though it showed a full charge before hand.

For your first question, I don't know the answer either and I hope someone else knows how to do it.

yoyojacky, 2024-04-09:
please  try UPS Plus

closed 2024-04-09 by yoyojacky