geeekpi/upsv3 #10: [EP-0118 UPS] Main power cuts off to battery

himuura, 2021-09-13:
Hey! I've bought an UPS SKU: EP-0118 and i've been trying it out. I found an issue that i did not like whatsoever...i usually restart my pi at 4am so i dont know if this issue stems from this.

First time testing the UPS with an official PSU for Rpi 3B (around 2.1A). After the first boot, everything was smooth (first 12 or so hours). Today, out of nowhere, when i woke up at 7.30am the Pi was completely off, despite having 2x Sony VTC6 batteries fully charged.
I don't know why, there wasn't a power off during the night since my other Rpi's were still functioning well and checking the logs, i can confirm that a power outage was not the issue.

So, in short, somehow during the night, the main power cut off and the batteries kicked in and stayed in battery mode until the energy of the batteries was depleted. I would like to know why.

I noticed one thing, in the morning, when i spotted the issue, clicked the power on button and the battery was full (according to the leds) but after 10s it went off. i tried a couple of times and the behaviour was the same. But after i unplugged, waited 5seconds and replugged, the led indicators suddenly showed it was depleted. This was strange...

Any way i can troubleshoot this issue?
THanks in advance!

MikeEitel, 2021-10-06:
I'm happen to have that 3 times. With 2 boards. This crap is in my view a systematic problem. And as I run an real ups as powersource there was surely not a power loss over night. There must be an design failure in the sw/hw.
I wrote to 52pi, but they do not react. I regret that I spend the money.

thiscantbeserious, 2022-01-08:
Same here with legit, very High Quality Samsung Li-Ion within the Specs.

Worked for about 24 hours, charged them to full - then suddenly its off. 

Restarting it didnt help, disconnected the Power-Supply to restart it again - this time only 1 Battery Level showing.

Nope, its not the Batteries its that "UPS" being badly designed. Completely unusuable in this state.

himuura, 2022-02-12:
Update on the topic. Got a 5V 5A PSU to connect to the UPS. Same thing happens...from time to time, out of nowhere, the battery gets depleted without warning. Been registering the current and voltage values with grafana, the lowest voltage i got was something like 5.084V. Don't know if it helps...another issue i spotted, despite being a 5V 5A (25W) PSU was that the batteries never got fully charged. The last led (75% =<) was never fully lit and always blinking. So my guess is most likely, as someone said, the UPS has some fault in the design and not interely usable. I do not recommend this device for sure. Gonna look for some alternatives...This issue can be closed.

19leva89, 2022-02-13:
У меня блок питания оригинальный на 3А, была та-же проблема, но после того как я подключил через переходник USB из комплекта, то все заработало отлично!
[image]

19leva89, 2022-02-13:
Блок питания на 3А шел в комплекте с Raspberry Pi. Сейчас у меня вообще стоит старый блок питания на 1А от телефона HTC и все отлично работает! Проблема решилась когда поставил адаптер из комплекта
[image]
Если это не помогает, то возможно поможет [перепрошивка платы ](https://github.com/geeekpi/upsv3/blob/master/UPS%20for%20RPi%204B%20firmware%20update%20tutorial.pdf)

himuura, 2022-02-13:
Well, in my case, that adapter did not come with the kit. I just got the board itself, no adapter inside. Any way i can buy it on aliexpress or amazon?

19leva89, 2022-02-13:
Я могу сделать фото адаптера с разных сторон, можете поискать на aliexpress. Или напишите производителю, где можно купить адаптер?

thiscantbeserious, 2022-02-17:
The seller gives a crap, neither the email, noir Github, noir Aliexpress they ever respond.

sebekhtc, 2023-12-11:
I have the adapter that came with the kit, it didnt help, the unit still switches off. Im going to change it with UPS HAT (B)

yoyojacky, 2024-04-09:
> The seller gives a crap, neither the email, noir Github, noir Aliexpress they ever respond.

Did you ever contact with sale man?  please contact our sale man for help.

closed 2024-04-09 by yoyojacky