geeekpi/upsv3 #14: Firmware update UPS has stopped working

ociekly, 2022-03-11:
Hello.
I had to update the firmware because my motherboard suffered from turning off after about 24 hours. For this purpose, I purchased the NU-Link programmer. I made the updates today according to the PDF instructions. Everything ended correctly

The system does not seem to be working. There is no reaction after installing the batteries. I press the switch, the charge indicator shows for a moment that the batteries are fully charged, but the power to the Raspberry is not turned on (the RPI POWER LED is off). Repeated presses also activate the indicator. Holding the key down also activates the indicator momentarily. After connecting the charger, the status indicator "fills up" and only the last LED flashes (batteries full). I tried to program the system for a few hours, but without success. Is the firmware file correct?

[image]

David

yoyojacky, 2024-04-09:
There is something wrong with your firmware. we do not recommend you update the firmware by yourself.

closed 2024-04-09 by yoyojacky

SplavSlav, 2024-10-07:
I have exactly the same problem as ociekly. How to fix it?

yoyojacky, 2024-10-21:
> 我遇到了与 ociekly 完全相同的问题。如何解决？

联系销售，提供你的order number，申请售后。